roboryman/prototroller #73: Rev-B Master and Module Board Design

evanzh15, 2023-03-01:
Make modifications to the Rev-A Master and Module Boards to adjust issues and design flaws in the original board.

Modify the following (but not limited to):

- 4x16 Decoder/Demux Footprint
- Change 10k Ohm resistor values to 1k
- Modify the 4 1x1 male header footprints into a single 1x4 male header footprint
- Change the SWD connector to a 1x4 male header footprint

evanzh15, 2023-03-01:
[https://github.com/roboryman/prototroller/tree/rev-b](url)

roboryman, 2023-03-01:
Added a bit. For both boards we should keep the 10k on the QSPI_NSS flash nets since our boards work great with them currently. We need to decide on what nets we are going to assign to the 4 extra pins on the mod boards.

closed 2023-03-07 by bcmcleavy